mathoudebine/turing-smart-screen-python #635: Showing "1 Jan 20254" after running during year change

sutaburosu, 2025-01-01:
**Describe the bug**  
System booted on 31 Dec now showing "1 Jan 20254" for the date. I'm doubtful that this is actually related to the year changing. I feel it's more likely that this happens on the 1st of every month when the date string shrinks by 1 character.

**To Reproduce**  
I haven't verified this to work, but presumably setting your system clock to a few moments before the new year/month change then starting this project would reproduce what I see.

**Expected behaviour**  
Display correct year.

**Screenshots / photos of the Turing screen**  
[image]

**Environment:**  
 - Turing 3.5"
 - commit 980577f4a3349a8041199acb0290f528401e6cc6 (from 21 Jul 2024)
 - Ubuntu 24.10
 - Python 3.12.7
 - AMD CPU + GPU

**Additional context**  
Add any other context about the problem here.

mathoudebine, 2025-01-02:
Hello, for the date this problem happens because it was not possible to guess a default size: it depends on the computer locale.

In your case, I think you can fix this by editing the theme and using the `MIN_SIZE` optional value that defines a minimum text size in characters:
[image]

It will move the date slightly to the right and keep en empty space at the beginning for 2-digits day numbers.

closed 2025-01-02 by mathoudebine

sutaburosu, 2025-01-02:
Nice one. Thank you for your continued dedication to this project. I'll upgrade now.