mathoudebine/turing-smart-screen-python #707: [AMD GPU] FPS and Temperature not showing

jpegCristo, 2025-03-05:
Hi, I recently installed the exe after using source for months. Already in the last period the display of FPS and GPU temperature had stopped working for me. Problem that now persists even with the exe.
My gpu is an RX7900 GRE. Does anyone know if there are any solutions? Or is it just a hardware incompatibility problem and therefore I have to wait?
Thank you in advance.

[image]

oopiicaa, 2025-03-26:
I have the same issue with executable version - previous version(s) work fine. Interesting thing is that one of the included themes seems to show my GPU data, but my other custom theme does not. I have compared config files of both themes and nothing seems different/out of the ordinary. I didn't have the time yet to dig deeper and since nobody else reported this problem, I assumed it was an issue on my side.

CPU data is fine. RX 7800 XT GPU.

mathoudebine, 2025-04-01:
Hello @oopiicaa @jpegCristo 
You can check this in your `theme.yaml`: the `INTERVAL` field is common to all GPU sensors and should be inside the `GPU` section e.g.:
``` yaml
  GPU:
    INTERVAL: 1
    PERCENTAGE:
      TEXT:
        SHOW: True
        SHOW_UNIT: True
        X: 35
        Y: 201
        FONT: jetbrains-mono/JetBrainsMono-Bold.ttf
        FONT_SIZE: 40
        FONT_COLOR: 255, 255, 255
        BACKGROUND_IMAGE: background.png
```
Unlike `CPU` where the interval is specific for each sensor

closed 2025-04-05 by mathoudebine