mathoudebine/turing-smart-screen-python #724: No info and scrambled screen

DonAlcasan, 2025-03-29:
**Describe the bug**  
8.8 inch turing screen not showing any numbers and scramble in the screen and there is no Turing Smart Screen icon in the system monitor.

**To Reproduce**  
Turn on the python program with python3 configure.py and hit "save and run"

**Expected behavior**  
Working correctly, this works fine when I use the screen in Windows 11

**Screenshots / photos of the Turing screen**  

[image]

[image]

**Environment:**  
 - Smart screen model: Turing 8.8
 - Revision of this project: 3.9.1
 - OS with version: Nobara Linux 41, Kernel 6.13.8
 - Python version Python 3.13.2
 - Hardware: AMD Ryzen 7950X3D CPU, AMD 9070XT GPU, 64 GB RAM Corsair

**Additional context**  
Output form terminal:

10:09:57 [DEBUG] HW revision: C
10:09:57 [DEBUG] Auto detected COM port: /dev/ttyACM1
10:09:57 [DEBUG] Using Python 3.13.2 (main, Feb  4 2025, 00:00:00) [GCC 14.2.1 20250110 (Red Hat 14.2.1-7)]
10:09:57 [INFO] Tray icon has been displayed
10:09:57 [INFO] Initialize display
10:09:57 [INFO] Display reset (COM port may change)...
10:10:12 [DEBUG] Static COM port: /dev/ttyACM1
10:10:12 [DEBUG] HW sub-revision returned: chs_88inch.dev1_rom1.90
10:10:12 [INFO] Calling ScreenOn
10:10:12 [DEBUG] Drawing Image: BACKGROUND
10:10:12 [DEBUG] Bitmap /home/alcasan/Apps/turingsmart/res/themes/AMD/background.png is now loaded in the cache
10:10:12 [DEBUG] Drawing Text: CPU
10:10:12 [DEBUG] Drawing Text: RAM
10:10:12 [DEBUG] Drawing Text: GPU
10:10:12 [DEBUG] Drawing Text: TIME
10:10:12 [INFO] Waiting for all pending request to be sent to display (10s max)...
10:10:13 [DEBUG] (Waited 0.3s)
10:10:13 [INFO] Starting system monitoring
10:10:14 [INFO] Detected AMD GPU(s)

youkai1981, 2025-03-31:
同样的问题

mathoudebine, 2025-04-01:
Hey @DonAlcasan @youkai1981 
I heard there is a new generation of Turing 8.8" smart screen.
Do you have a link to the original software provided with the screen?

DonAlcasan, 2025-04-01:
> Hey [@DonAlcasan](https://github.com/DonAlcasan) [@youkai1981](https://github.com/youkai1981) I heard there is a new generation of Turing 8.8" smart screen. Do you have a link to the original software provided with the screen?

Hi!

This is the one I bought in October 2024: https://www.aliexpress.com/item/1005005538830549.html?spm=a2g0o.order_list.order_list_main.243.7a4b1802BGmLyn
And the link that was provided is: https://lgb123-1253504678.cos.ap-beijing.myqcloud.com/88inchENG.rar

And thank you for making this, I've gotten everything to work since moving completely to Linux except this screen =)

youkai1981, 2025-04-01:
> 嘿[@DonAlcasan](https://github.com/DonAlcasan) [@youkai1981](https://github.com/youkai1981) 听说现在有新一代的图灵8.8寸智能屏， 请问有原装随屏软件的链接吗？
网站是
https://www.turzx.com/2023/03/02/%e7%9b%b4%e9%93%be%e4%b8%8b%e8%bd%bddirectdownload/
下载连接
https://lgb123-1253504678.cos.ap-beijing.myqcloud.com/88inch.rar

youkai1981, 2025-04-01:
> Hey [@DonAlcasan](https://github.com/DonAlcasan) [@youkai1981](https://github.com/youkai1981) I heard there is a new generation of Turing 8.8" smart screen. Do you have a link to the original software provided with the screen?

网站是
https://www.turzx.com/2023/03/02/%e7%9b%b4%e9%93%be%e4%b8%8b%e8%bd%bddirectdownload/
下载连接
https://lgb123-1253504678.cos.ap-beijing.myqcloud.com/88inch.rar

youkai1981, 2025-04-01:
本项目与屏幕自带的软件对比[nvdia-o.txt](https://github.com/user-attachments/files/19556835/nvdia-o.txt)

[nvdia-n.txt](https://github.com/user-attachments/files/19556840/nvdia-n.txt)

archibaldodlc, 2025-04-04:
Hi! First of all, thanks for this project, it´s amazing.

I installed on both ways, with venv and from the OS and have the same problem report by @DonAlcasan.

My system:

Kubuntu 24.10
Kernel: 6.11.0-21-generic
Python 3.12.7
Hardware: Intel 14900k, Nvidia 4080GPU, 128 GB RAM Hyper X.
Revision: Main from Git repository.

The symptons are the same, no data showing and scramble in monitor. 

I´m available to test if it´s needed.

Thanks again!

mathoudebine, 2025-04-05:
@archibaldodlc @youkai1981 @DonAlcasan Thank you for your feedback. It seems here your screen is supported by this program.
I need additional information from you to help me debug this issue:

- What theme do you use?
- Did you configure "Orientation" as "classic" or "reverse"?
- When you run the System Monitor, you should have a line in the terminal or in "logs.txt" file that says : `[DEBUG] HW sub-revision returned: chs_88inch.dev1_rom1.88` with the display firmware version. Can you show me this line please?

mathoudebine, 2025-04-05:
Also, I tried to implement a fix on this branch `fix/724-turing-88-no-info-and-scrambled-screen` if you have the time to try it on your side and tell me if it fixes the issue

archibaldodlc, 2025-04-07:
Hi, @mathoudebine, thanks for your response. 

I tried the fix, but sadly didn´t work. 

This are the settings that I used:

Settings:

[image]

Logs:

```
~/g/1/turing-smart-screen-python fix/724-turi…mbled-screen !1 ?1 ❯ 
07/04/25 14:56:09 [INFO] Loading theme AMD from res/themes/AMD/theme.yaml
07/04/25 14:56:09 [DEBUG] HW revision: C
07/04/25 14:56:09 [DEBUG] Auto detected COM port: /dev/ttyACM1
07/04/25 14:56:09 [DEBUG] Using Python 3.12.7 (main, Feb  4 2025, 14:46:03) [GCC 14.2.0]
07/04/25 14:56:09 [INFO] Tray icon has been displayed
07/04/25 14:56:09 [INFO] Initialize display
07/04/25 14:56:09 [INFO] Display reset (COM port may change)...
07/04/25 14:56:24 [DEBUG] Static COM port: /dev/ttyACM1
07/04/25 14:56:24 [DEBUG] HW sub-revision returned: chs_88inch.dev1_rom1.90
07/04/25 14:56:24 [INFO] Calling ScreenOn
07/04/25 14:56:24 [DEBUG] Drawing Image: BACKGROUND
07/04/25 14:56:24 [DEBUG] Bitmap /home/archibaldo/git/10-OtrasCosas/turing-smart-screen-python/res/themes/AMD/background.png is now loaded in the cache
07/04/25 14:56:24 [DEBUG] Drawing Text: CPU
07/04/25 14:56:24 [DEBUG] Drawing Text: RAM
07/04/25 14:56:24 [DEBUG] Drawing Text: GPU
07/04/25 14:56:24 [DEBUG] Drawing Text: TIME
07/04/25 14:56:24 [INFO] Waiting for all pending request to be sent to display (10s max)...
07/04/25 14:56:24 [DEBUG] (Waited 0.3s)
07/04/25 14:56:24 [INFO] Starting system monitoring
07/04/25 14:56:26 [INFO] Detected Nvidia GPU(s)

```

A small video:

https://github.com/user-attachments/assets/c2637d04-3ec7-4012-9694-edc12cefaa37

nreinartz, 2025-04-22:
ROM Version 1.89 uses a different image encoding when transfering the image changes. This is most likely the cause for the artifacts.

manijbasnet, 2025-04-24:
@mathoudebine Thank you for your work in this repo - I am learning a lot. Unfortunately, I am having same issue with same log as @archibaldodlc. Let me know if you would like me to test anything. I am using Fedora Workstation 42.

@nreinartz I am new to this. Does this mean we need something from turzx themselves or is there something we can do to fix?

nreinartz, 2025-04-24:
@manijbasnet The lcd_comm_rev_c needs to be adjusted. 

1. The ROM Version needs to be parsed (response of Hello Command)
2. The pixel format of the image needs to be changed based on what ROM version is used

>=1.89 = RGBA (4 Bytes)
< 1.89 = BGR + Compressed Alpha channel (3 Bytes)

I also got one with ROM version 1.90 and can confirm that this works.

manijbasnet, 2025-04-24:
@nreinartz that's awesome you got it working. I set my rom version to `chs_88inch.dev1_rom1.90` and removed the conversion  to bgra. no more scrambled bits but all i still can't see the texts and color have changed :(

[image]

nreinartz, 2025-04-24:
@manijbasnet I don't know what changes you made. You can use my working branch (\>= 1.89 only) which I worked on some months ago. The important stuff is at the bottom of the function that generates the updates. You can try to adapt it to the current version.

https://github.com/nreinartz/turing-smart-screen-python/blob/main/library/lcd/lcd_comm_rev_c.py

mathoudebine, 2025-05-16:
Hello @manijbasnet @archibaldodlc @DonAlcasan 
Once again I need your help to test the latest commits on branch `fix/724-turing-88-no-info-and-scrambled-screen`
I added the RGBA mode for Turing 8.8" based on @nreinartz code and advice
Unfortunately I cannot test it myself because it is not possible anymore to buy Turing 8.8" with ROM 1.90 (there is a new hardware revision that has replaced it in stores)
Please give me your feedback here as well as the logs, thanks

nreinartz, 2025-05-16:
I had a look at my python code and while it worked with my v1.90 display, it's still not 100% correct. I am working on my own solution to connect with the displays, so I can only provide C# code, but the correct way to transfer the pixels would look like this @mathoudebine :

```cs
if (_romVersion >= 1.89) {
    dataSpan[dataIndex++] = bitmapData[pixelIndex];
    dataSpan[dataIndex++] = bitmapData[pixelIndex + 1];
    dataSpan[dataIndex++] = bitmapData[pixelIndex + 2];
    dataSpan[dataIndex++] = bitmapData[pixelIndex + 3];
} else {
    byte b = bitmapData[pixelIndex];
    byte g = bitmapData[pixelIndex + 1];
    byte r = bitmapData[pixelIndex + 2];
    byte a = bitmapData[pixelIndex + 3];

    int alphaHighNibble = a >> 4;
    dataSpan[dataIndex++] = (byte)(b & 0xFC | alphaHighNibble >> 2);
    dataSpan[dataIndex++] = (byte)(g & 0xFC | alphaHighNibble & 3);
    dataSpan[dataIndex++] = r;
}
```

Both versions transfer an alpha channel, but <1.89 compresses it. This also enables us to play a video behind the theme, since the alpha channels are now correctly transmitted. There might also be 8.8 Inch versions that come with ROM Version 1.89, so checking for 1.90 might not be enough.

mathoudebine, 2025-05-16:
Thanks for the feedback @nreinartz I wasn't aware of that! 
Out of curiosity, where did you find all this info? If there is some kind of datasheet I would be interested

nreinartz, 2025-05-16:
Decompiled, deobfuscated & debugged the Turzx application @mathoudebine

mathoudebine, 2025-05-16:
Nice work! Did you use a specific tool for deobfuscating? I previously tried to decompile the software but was stuck with the obfuscation

archibaldodlc, 2025-05-18:
It works!  I tested right now and it's working. I attach a video 

https://github.com/user-attachments/assets/772c9d3d-8402-4131-9597-87fe2eed81f3

No scramble and the stats works correctly. Thank you, @mathoudebine!

manijbasnet, 2025-05-18:
It works for me too. Demo on a theme i am currently working on.

https://github.com/user-attachments/assets/d4c8da0e-ae37-4f0e-9959-9aac8b105e45

```18/05/25 12:29:05 [INFO] Loading theme SimpleFuturePurpleVertical from res/themes/SimpleFuturePurpleVertical/theme.yaml
18/05/25 12:29:05 [DEBUG] HW revision: C
18/05/25 12:29:05 [DEBUG] Static COM port: /dev/ttyACM1
18/05/25 12:29:05 [DEBUG] Using Python 3.11.12 (main, Apr 22 2025, 17:36:53) [GCC 15.0.1 20250329 (Red Hat 15.0.1-0)]
18/05/25 12:29:05 [INFO] Tray icon has been displayed
18/05/25 12:29:05 [INFO] Initialize display
18/05/25 12:29:05 [INFO] Display reset (COM port may change)...
18/05/25 12:29:05 [ERROR] Failed to dock icon
Traceback (most recent call last):
  File "/home/manijbasnet/.pyenv/versions/turzx-screen/lib/python3.11/site-packages/pystray/_xorg.py", line 162, in _update_icon
    self._assert_docked()
  File "/home/manijbasnet/.pyenv/versions/turzx-screen/lib/python3.11/site-packages/pystray/_xorg.py", line 395, in _assert_docked
    assert self._systray_manager
           ^^^^^^^^^^^^^^^^^^^^^
AssertionError
18/05/25 12:29:05 [ERROR] Failed to dock icon
Traceback (most recent call last):
  File "/home/manijbasnet/.pyenv/versions/turzx-screen/lib/python3.11/site-packages/pystray/_xorg.py", line 143, in _show
    self._assert_docked()
  File "/home/manijbasnet/.pyenv/versions/turzx-screen/lib/python3.11/site-packages/pystray/_xorg.py", line 395, in _assert_docked
    assert self._systray_manager
           ^^^^^^^^^^^^^^^^^^^^^
AssertionError
18/05/25 12:29:20 [DEBUG] Static COM port: /dev/ttyACM1
18/05/25 12:29:20 [DEBUG] Display ID returned: chs_88inch.dev1_rom1.90
18/05/25 12:29:20 [DEBUG] HW sub-revision detected: SubRevision.REV_8INCH, ROM version: 90
18/05/25 12:29:20 [DEBUG] Drawing Image: BACKGROUND
18/05/25 12:29:20 [DEBUG] Bitmap /home/manijbasnet/Workspace/turing-smart-screen-python/res/themes/SimpleFuturePurpleVertical/background.png is now loaded in the cache
18/05/25 12:29:20 [DEBUG] Drawing Text: CPU_MODEL
18/05/25 12:29:20 [DEBUG] Drawing Text: CPU_MEMORY
18/05/25 12:29:20 [DEBUG] Drawing Text: GPU_MODEL
18/05/25 12:29:20 [DEBUG] Drawing Text: GPU_MEMORY
18/05/25 12:29:20 [DEBUG] Drawing Text: GPU_FPS
18/05/25 12:29:20 [INFO] Waiting for all pending request to be sent to display (10s max)...
18/05/25 12:29:20 [DEBUG] (Waited 0.3s)
18/05/25 12:29:20 [INFO] Starting system monitoring
18/05/25 12:29:21 [WARNING] Your CPU Fan sensor could not be auto-detected. Select it from Configuration UI.
18/05/25 12:29:21 [INFO] Detected AMD GPU(s)
18/05/25 12:29:21 [WARNING] Your GPU FPS is not supported yet
18/05/25 12:29:21 [WARNING] Your GPU Fan Speed is not supported yet
```

Thanks for the fix @mathoudebine @nreinartz. Happy to assist in any future testing.

dhegi-lab, 2025-05-18:
Hello,

I can confirm that!
I have just tested it and it works perfectly.

Thank you very much!

Best regards

DonAlcasan, 2025-05-18:
Can confirm, the fix works and everything seem to work. good job! thank you!

DonAlcasan, 2025-05-21:
> It works for me too. Demo on a theme i am currently working on.
> 
>  IMG_5985_480.mov 
> ```
> 18/05/25 12:29:05 [DEBUG] HW revision: C
> 18/05/25 12:29:05 [DEBUG] Static COM port: /dev/ttyACM1
> 18/05/25 12:29:05 [DEBUG] Using Python 3.11.12 (main, Apr 22 2025, 17:36:53) [GCC 15.0.1 20250329 (Red Hat 15.0.1-0)]
> 18/05/25 12:29:05 [INFO] Tray icon has been displayed
> 18/05/25 12:29:05 [INFO] Initialize display
> 18/05/25 12:29:05 [INFO] Display reset (COM port may change)...
> 18/05/25 12:29:05 [ERROR] Failed to dock icon
> Traceback (most recent call last):
>   File "/home/manijbasnet/.pyenv/versions/turzx-screen/lib/python3.11/site-packages/pystray/_xorg.py", line 162, in _update_icon
>     self._assert_docked()
>   File "/home/manijbasnet/.pyenv/versions/turzx-screen/lib/python3.11/site-packages/pystray/_xorg.py", line 395, in _assert_docked
>     assert self._systray_manager
>            ^^^^^^^^^^^^^^^^^^^^^
> AssertionError
> 18/05/25 12:29:05 [ERROR] Failed to dock icon
> Traceback (most recent call last):
>   File "/home/manijbasnet/.pyenv/versions/turzx-screen/lib/python3.11/site-packages/pystray/_xorg.py", line 143, in _show
>     self._assert_docked()
>   File "/home/manijbasnet/.pyenv/versions/turzx-screen/lib/python3.11/site-packages/pystray/_xorg.py", line 395, in _assert_docked
>     assert self._systray_manager
>            ^^^^^^^^^^^^^^^^^^^^^
> AssertionError
> 18/05/25 12:29:20 [DEBUG] Static COM port: /dev/ttyACM1
> 18/05/25 12:29:20 [DEBUG] Display ID returned: chs_88inch.dev1_rom1.90
> 18/05/25 12:29:20 [DEBUG] HW sub-revision detected: SubRevision.REV_8INCH, ROM version: 90
> 18/05/25 12:29:20 [DEBUG] Drawing Image: BACKGROUND
> 18/05/25 12:29:20 [DEBUG] Bitmap /home/manijbasnet/Workspace/turing-smart-screen-python/res/themes/SimpleFuturePurpleVertical/background.png is now loaded in the cache
> 18/05/25 12:29:20 [DEBUG] Drawing Text: CPU_MODEL
> 18/05/25 12:29:20 [DEBUG] Drawing Text: CPU_MEMORY
> 18/05/25 12:29:20 [DEBUG] Drawing Text: GPU_MODEL
> 18/05/25 12:29:20 [DEBUG] Drawing Text: GPU_MEMORY
> 18/05/25 12:29:20 [DEBUG] Drawing Text: GPU_FPS
> 18/05/25 12:29:20 [INFO] Waiting for all pending request to be sent to display (10s max)...
> 18/05/25 12:29:20 [DEBUG] (Waited 0.3s)
> 18/05/25 12:29:20 [INFO] Starting system monitoring
> 18/05/25 12:29:21 [WARNING] Your CPU Fan sensor could not be auto-detected. Select it from Configuration UI.
> 18/05/25 12:29:21 [INFO] Detected AMD GPU(s)
> 18/05/25 12:29:21 [WARNING] Your GPU FPS is not supported yet
> 18/05/25 12:29:21 [WARNING] Your GPU Fan Speed is not supported yet
> ```
> 
> Thanks for the fix [@mathoudebine](https://github.com/mathoudebine) [@nreinartz](https://github.com/nreinartz). Happy to assist in any future testing.

Hi

That theme looks great, i've been making a few of my own after I got it working but I don't know how to make a radial circle like that with steps like that, can you share the theme.yaml file so I can learn ?
Sorry for posting this here, didn't know how to contact him otherwise.

mathoudebine, 2025-05-21:
Many thanks to all of you for taking the time to test it, the fix has been merged and will be included in the next release!

manijbasnet, 2025-06-01:
> Hi
>
> That theme looks great, i've been making a few of my own after I got it working but I don't know how to make a radial circle like that with steps like that, can you share the theme.yaml file so [...]

Hi @DonAlcasan . Have a look at this commit I made on a fork. https://github.com/manijbasnet/turing-smart-screen-python/commit/9ffc99a02197bfa33c0c25942b218be6ccd63a84 you will need few other changes to get the theme to work as seen in the video.