mathoudebine/turing-smart-screen-python #731: Mis

tapsu01, 2025-04-05:
**Describe the bug**  
when i try to add pyamdgpuinfo, I am getting a error `'xf86drm.h' file not found`

**To Reproduce**  
Steps to reproduce the behavior:
1. Run ```sh 
python3 -m pip install pyamdgpuinfo
```
2. See error

**Environment:**  
 - Smart screen model: Turing 3.5"
 - Revision of this project: 3.9.2 
 - OS with version:  MacOS 13.7
 - Python version: 3.12.7
 - Hardware [Intel CPU core i7 8700, AMD Radeon Pro WX 4100 4 GB]

**Screenshots / photos of the Turing screen**  
```sh
python3 -m pip install pyamdgpuinfo
Collecting pyamdgpuinfo
  Using cached pyamdgpuinfo-2.1.6.tar.gz (21 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: pyamdgpuinfo
  Building wheel for pyamdgpuinfo (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for pyamdgpuinfo (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [27 lines of output]
      /private/var/folders/14/6qcc7jl95h9dkqcg2hz09_dr0000gn/T/pip-build-env-pw389bel/overlay/lib/python3.12/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
      !!

              ********************************************************************************
              Please consider removing the following classifiers in favor of a SPDX license expression:

              License :: OSI Approved :: GNU General Public License v3 (GPLv3)

              See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
              ********************************************************************************

      !!
        self._finalize_license_expression()
      running bdist_wheel
      running build
      running build_py
      creating build/lib.macosx-10.13-universal2-cpython-312/pyamdgpuinfo
      copying pyamdgpuinfo/__init__.py -> build/lib.macosx-10.13-universal2-cpython-312/pyamdgpuinfo
      running build_ext
      building 'pyamdgpuinfo._pyamdgpuinfo' extension
      creating build/temp.macosx-10.13-universal2-cpython-312/pyamdgpuinfo
      clang -fno-strict-overflow -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -O3 -Wall -arch arm64 -arch x86_64 -g -I/usr/local/opt/ruby/include -I/usr/include/libdrm -I/Library/Frameworks/Python.framework/Versions/3.12/include/python3.12 -c pyamdgpuinfo/_pyamdgpuinfo.c -o build/temp.macosx-10.13-universal2-cpython-312/pyamdgpuinfo/_pyamdgpuinfo.o
      pyamdgpuinfo/_pyamdgpuinfo.c:1241:10: fatal error: 'xf86drm.h' file not found
      #include "xf86drm.h"
               ^~~~~~~~~~~
      1 error generated.
      error: command '/usr/bin/clang' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyamdgpuinfo
Failed to build pyamdgpuinfo
ERROR: Failed to build installable wheels for some pyproject.toml based projects (pyamdgpuinfo)
```

mathoudebine, 2025-04-05:
Hello @tapsu01 
Sorry but the pyamdgpuinfo package is only for Linux platforms, as you can see here on the requirements file:
https://github.com/mathoudebine/turing-smart-screen-python/blob/b3f08b8bcaf48ea57ac4dd847457dce92806e88b/requirements.txt#L25-L26
There was a mistake in the troubleshooting page, I fixed it.

For now I didn't find a Python library to get AMD GPU info on MacOS platforms. If you find one please let me know and I will try to integrate it to this project.

tapsu01, 2025-04-05:
Hello @mathoudebine 
So does that mean I can't currently use the Turing Smart Screen 3.5" on macOS and with an AMD GPU? In that case, I guess I'll have to let the Turing Smart Screen go to waste.