mathoudebine/turing-smart-screen-python #732: 3.9.2 Nvidia GPU stats not loading - Exception in GPU_Stats KeyError: 'MEMORY_TOTAL'

benjaminfenech, 2025-04-05:
**Describe the bug**  
GPU current memory, total memory, temperature, and line graph failing to be retrieved and drawn. Seeing an exception thrown GPU_Stats in the log output

**To Reproduce**  
Steps to reproduce the behavior:
1. Run configure.py or main.py
2. System monitoring starts, CPU, System Memory, and Networking stats are drawn to screen.
3. An exception is thrown in GPU_Stats, no GPU statistics are drawn to screen
```05/04/25 18:14:04 [INFO] Waiting for all pending request to be sent to display (10s max)...
05/04/25 18:14:06 [DEBUG] (Waited 1.9s)
05/04/25 18:14:06 [INFO] Starting system monitoring
05/04/25 18:14:07 [INFO] Detected Nvidia GPU(s)
Exception in thread GPU_Stats:
Traceback (most recent call last):
  File "/usr/lib/python3.13/threading.py", line 1041, in _bootstrap_inner
    self.run()
    ~~~~~~~~^^
  File "/usr/lib/python3.13/threading.py", line 992, in run
    self._target(*self._args, **self._kwargs)
    ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/benjamin/Documents/turing-smart-screen-python-3.9.2/library/scheduler.py", line 75, in wrap
    periodic(scheduler, interval, func)
    ~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/benjamin/Documents/turing-smart-screen-python-3.9.2/library/scheduler.py", line 64, in periodic
    action(*actionargs)
    ~~~~~~^^^^^^^^^^^^^
  File "/home/benjamin/Documents/turing-smart-screen-python-3.9.2/library/scheduler.py", line 128, in GpuStats
    stats.Gpu.stats()
    ~~~~~~~~~~~~~~~^^
  File "/home/benjamin/Documents/turing-smart-screen-python-3.9.2/library/stats.py", line 488, in stats
    gpu_mem_total_text_data = theme_gpu_data['MEMORY_TOTAL']['TEXT']
                              ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
KeyError: 'MEMORY_TOTAL'
```

**Expected behavior**  
No exception is thrown, and gpu stats are drawn to screen.

**Environment:**  
 - Smart screen model      - Turing 3.5
 - Revision of this project - 3.9.2 release
 - OS with version              - Kubuntu 25.04
 - Python version               - Python 3.13
 - Hardware                        - Intel i7 13700k, Nvidia RTX 4070ti

**Additional context**  
Did upgrade to Kubuntu 25.04 beta last night, everything seems fine so far, but had issues with pip not being recognised in my existing venv. Have downloaded latest release and created a new venv, everything has installed without errors - only seeing the GPU stat errors when trying to launch.
Full output of starting main.py, it's using a theme I found on forums, but I'm getting gpu stat errors regardless of theme chosen:
```05/04/25 18:13:58 [INFO] Loading theme Classic black from res/themes/Classic black/theme.yaml
05/04/25 18:13:58 [WARNING] Cannot find valid DISPLAY_SIZE property in selected theme Classic black, defaulting to 3.5"
05/04/25 18:13:58 [DEBUG] HW revision: A
05/04/25 18:13:58 [DEBUG] Auto detected COM port: /dev/ttyACM0
05/04/25 18:13:58 [DEBUG] Using Python 3.13.2 (main, Mar 29 2025, 10:04:43) [GCC 14.2.0]
05/04/25 18:13:58 [INFO] Tray icon has been displayed
05/04/25 18:13:58 [INFO] Initialize display
05/04/25 18:13:58 [INFO] Display reset (COM port may change)...
05/04/25 18:14:03 [DEBUG] Static COM port: /dev/ttyACM0
05/04/25 18:14:04 [DEBUG] HW sub-revision: SubRevision.TURING_3_5
05/04/25 18:14:04 [DEBUG] Drawing Image: BACKGROUND
05/04/25 18:14:04 [DEBUG] Bitmap /home/benjamin/Documents/turing-smart-screen-python-3.9.2/res/themes/Classic black/background.png is now loaded in the cache
05/04/25 18:14:04 [DEBUG] Drawing Text: TEXT_CPU
05/04/25 18:14:04 [DEBUG] Drawing Text: TEXT_GPU
05/04/25 18:14:04 [DEBUG] Drawing Text: TEXT_RAM
05/04/25 18:14:04 [DEBUG] Drawing Text: TEXT_RAM/
05/04/25 18:14:04 [DEBUG] Drawing Text: TEXT_DOWN
05/04/25 18:14:04 [DEBUG] Drawing Text: TEXT_UP
05/04/25 18:14:04 [INFO] Waiting for all pending request to be sent to display (10s max)...
05/04/25 18:14:06 [DEBUG] (Waited 1.9s)
05/04/25 18:14:06 [INFO] Starting system monitoring
05/04/25 18:14:07 [INFO] Detected Nvidia GPU(s)
Exception in thread GPU_Stats:
Traceback (most recent call last):
  File "/usr/lib/python3.13/threading.py", line 1041, in _bootstrap_inner
    self.run()
    ~~~~~~~~^^
  File "/usr/lib/python3.13/threading.py", line 992, in run
    self._target(*self._args, **self._kwargs)
    ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/benjamin/Documents/turing-smart-screen-python-3.9.2/library/scheduler.py", line 75, in wrap
    periodic(scheduler, interval, func)
    ~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/benjamin/Documents/turing-smart-screen-python-3.9.2/library/scheduler.py", line 64, in periodic
    action(*actionargs)
    ~~~~~~^^^^^^^^^^^^^
  File "/home/benjamin/Documents/turing-smart-screen-python-3.9.2/library/scheduler.py", line 128, in GpuStats
    stats.Gpu.stats()
    ~~~~~~~~~~~~~~~^^
  File "/home/benjamin/Documents/turing-smart-screen-python-3.9.2/library/stats.py", line 488, in stats
    gpu_mem_total_text_data = theme_gpu_data['MEMORY_TOTAL']['TEXT']
                              ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
KeyError: 'MEMORY_TOTAL'
^C05/04/25 18:32:11 [INFO] Caught signal 2, exiting
05/04/25 18:32:11 [INFO] Waiting for all pending request to be sent to display (5s max)...
05/04/25 18:32:12 [DEBUG] (Waited 0.5s)
```
I checked log.log but it isn't showing any of these errors after start system monitoring and detected nvidia gpu.

mathoudebine, 2025-04-05:
Hi @benjaminfenech 
I can't reproduce the issue on my side, did you copy the `res/themes` folder from an older version?
If yes, then you still need to keep the `res/themes/default.yaml` file content aligned with the new version. You can download it here for the 3.9.2: https://github.com/mathoudebine/turing-smart-screen-python/blob/3.9.2/res/themes/default.yaml

benjaminfenech, 2025-04-05:
Hey @mathoudebine 
Yeah I just copied whole res directory over because I thought it would be easier, didn't even think about it causing issues. 

I  have restored the 3.9.2 res folder and copied the individual theme and font over and it is now working as intended.

Thanks for your help :)

closed 2025-04-05 by benjaminfenech