mathoudebine/turing-smart-screen-python #761: video support for 3.5" turing smart display

offoxx, 2025-05-05:
hello,

Can you add a video background in 3.5" turing smart display?

thanks in advance.

Mumrikken, 2025-05-05:
Not possible unfortunatly. The video support on the 5, 2.1 and 8.8 inch displays works by uploading the video to an internal SD card on the device, and playing it from there. The 3.5 inch, to my knowledge, only supports sending images over USB. If you want video, you have to pick up another device i think, unless the one you bought spesifically mentions having video support, which i havent seen yet.

nreinartz, 2025-05-05:
One could try to do it like the 8.8 Rev 1.1 and just send each video frame separately instead of uploading the video, but I doubt that the bandwidth is sufficient. I'll test later how many frames per second can be sent.