mathoudebine/turing-smart-screen-python #769: Compatibility with this screen (software says V4)

LoLo2207, 2025-05-15:
Hi, I just got a screen that I thought it was the Turing 5" or other compatible with this project.

This is the listing from Aliexpress:

https://www.aliexpress.com/item/1005007391882540.html?spm=a2g0o.order_list.order_list_main.17.2f151802luNCt2

<details>
<summary>
And in case it gets removed, i'll paste some images from it:</summary>

[image]

[image]

</details>

At first I tried running this project but it didn't work. I get this output after running the configure script and clicking Save & Run

[image]

[image]

[image]

After that I tried to run the 5" and 3.5" software from turing https://www.turzx.com/2023/03/02/%E7%9B%B4%E9%93%BE%E4%B8%8B%E8%BD%BDdirectdownload/ but these didn't work either.

<details>
<summary>
I tried looking at the board to see if it was any from your wiki, but it's not.
</summary>

[image]

[image]

</details>

After reading the Aliexpress listing, this is the software that works. It's called **5 inch SmartMonitor V4** [https://smartdisplay.lanzouo.com/iGWBl2avuzod](https://smartdisplay.lanzouo.com/iGWBl2avuzod?spm=a2g0o.detail.1000023.1.127cZWRDZWRDoW) 

[image]

Is it possible to use it with this project?

Thanks.

mathoudebine, 2025-05-16:
Hello! Based on the software link from https://smartdisplay.lanzouo.com, I would say it is the 5" version of the Fuldho 3.5" IPS Screen that is not supported (see the "not supported" section from the front page) but there is still a chance: are you sure that COM3 is your screen?
To confirm, if you disconnect the screen and restart the `configure.py` program, is COM3 still listed in the com ports?

nreinartz, 2025-05-16:
I also bought one of these thinking it was a Turing screen. It seems like the rendering on these works in a completely different way. The themes seem to be rendered on the device itself, not with the software. The software only sends the changed sensor values to the display. At least that's what I gathered when analyzing the usb traffic. To make matters worse, it is C++ and therefore not as easy to reverse engineer as the Turzx software.

LoLo2207, 2025-05-16:
> Hello! Based on the software link from https://smartdisplay.lanzouo.com, I would say it is the 5" version of the Fuldho 3.5" IPS Screen that is not supported (see the "not supported" section from [...]

Hi, I'm sure it's the COM3 port.

What a bummer, I'll try to return it or sell it and get a Turing one, thanks.

closed 2025-05-22 by LoLo2207