mathoudebine/turing-smart-screen-python #776: Cannot run configure.py on Mac

flodi, 2025-05-23:
After installing all the dependecies without errors, running

python3 configure.py

result in

flodi@fWork turing-smart-screen-python-3.9.3 % python3 ./configure.py
DEPRECATION WARNING: The system version of Tk is deprecated and may be removed in a future release. Please don't rely on it. Set TK_SILENCE_DEPRECATION=1 to suppress this warning.
Traceback (most recent call last):
  File "/Users/flodi/Downloads/turing-smart-screen-python-3.9.3/./configure.py", line 686, in <module>
    configurator = TuringConfigWindow()
  File "/Users/flodi/Downloads/turing-smart-screen-python-3.9.3/./configure.py", line 181, in __init__
    self.window.iconphoto(True, PhotoImage(file=MAIN_DIRECTORY + "res/icons/monitor-icon-17865/64.png"))
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/tkinter/__init__.py", line 4064, in __init__
    Image.__init__(self, 'photo', name, cnf, master, **kw)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/tkinter/__init__.py", line 4009, in __init__
    self.tk.call(('image', 'create', imgtype, name,) + options)
_tkinter.TclError: couldn't recognize data in image file "/Users/flodi/Downloads/turing-smart-screen-python-3.9.3/res/icons/monitor-icon-17865/64.png"
flodi@fWork turing-smart-screen-python-3.9.3 %

I'm on MacOS 15.3.2 with homebrew 4.5.2

mathoudebine, 2025-05-23:
Thanks for raising this issue, it seems you may have a version of tkinter on your system that does not support PNG decoding. Can you try the following fix for me?

In `turing-smart-screen-python-3.9.3/configure.py` replace the line 181:
```python
self.window.iconphoto(True, PhotoImage(file=MAIN_DIRECTORY + "res/icons/monitor-icon-17865/64.png"))
```
by
```python
self.window.iconphoto(True, ImageTk.PhotoImage(Image.open(MAIN_DIRECTORY + "res/icons/monitor-icon-17865/64.png")))
```

flodi, 2025-05-23:
Thank you for the fast answer.

I get this

flodi@fWork turing-smart-screen-python-3.9.3 % python3 ./configure.py
DEPRECATION WARNING: The system version of Tk is deprecated and may be removed in a future release. Please don't rely on it. Set TK_SILENCE_DEPRECATION=1 to suppress this warning.
Traceback (most recent call last):
  File "/Users/flodi/Downloads/turing-smart-screen-python-3.9.3/./configure.py", line 687, in <module>
    configurator = TuringConfigWindow()
  File "/Users/flodi/Downloads/turing-smart-screen-python-3.9.3/./configure.py", line 188, in __init__
    self.more_config_window = MoreConfigWindow(self)
  File "/Users/flodi/Downloads/turing-smart-screen-python-3.9.3/./configure.py", line 563, in __init__
    sv_ttk.set_theme("light")
  File "/Users/flodi/Library/Python/3.9/lib/python/site-packages/sv_ttk/__init__.py", line 30, in set_theme
    _load_theme(style)
  File "/Users/flodi/Library/Python/3.9/lib/python/site-packages/sv_ttk/__init__.py", line 16, in _load_theme
    style.tk.call("source", str(TCL_THEME_FILE_PATH))
_tkinter.TclError: version conflict for package "Tk": have 8.5.9, need 8.6

II've installed just now Tk with brew along your software.

mathoudebine, 2025-05-23:
What command did you use to install Python and tk? I'm surprised you have a quite old version of Python (3.9) and even older TK (8.5.9) unless it is what you wanted?

I recommend to install a much more recent version like https://formulae.brew.sh/formula/python-tk@3.13 which contains both Python 3.13.3 and TK 9.0.1

flodi, 2025-05-23:
Ok, there was the default MacOs python messing around, now I have the right versions

But I get

(miniscreen) flodi@fWork turing-smart-screen-python-3.9.3 % python3 configure.py
Traceback (most recent call last):
  File "/Users/flodi/Downloads/turing-smart-screen-python-3.9.3/configure.py", line 687, in <module>
    configurator = TuringConfigWindow()
  File "/Users/flodi/Downloads/turing-smart-screen-python-3.9.3/configure.py", line 188, in __init__
    self.more_config_window = MoreConfigWindow(self)
                              ~~~~~~~~~~~~~~~~^^^^^^
  File "/Users/flodi/Downloads/turing-smart-screen-python-3.9.3/configure.py", line 563, in __init__
    sv_ttk.set_theme("light")
    ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/Users/flodi/miniscreen/lib/python3.13/site-packages/sv_ttk/__init__.py", line 30, in set_theme
    _load_theme(style)
    ~~~~~~~~~~~^^^^^^^
  File "/Users/flodi/miniscreen/lib/python3.13/site-packages/sv_ttk/__init__.py", line 16, in _load_theme
    style.tk.call("source", str(TCL_THEME_FILE_PATH))
    ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
_tkinter.TclError: version conflict for package "Tk": have 9.0.1, need 8.6

flodi, 2025-06-12:
I was using /usr/local to install, moved to user home all worked

Now I get

No supported GPU found

I'm using a Mac Mini M4

l2fprod, 2026-01-03:
just got the screen and simple-program.py worked perfectly on Mac Mini M4

now going to play with the API -- thank you for the project!