mathoudebine/turing-smart-screen-python #807: OpenWeatherMap :  Error 401

DroidBV8, 2025-07-14:
**Describe the bug**  
**To Reproduce**  

Unable to display OpenWeatherMap data. 
I have created account and free api key and config these on the Turing System monitor App

I get the following message : Please NOTE THAT USIN

Theme USED: CyberArasaka - Landscape

**Expected behavior**  
I would like OpenWeatherMap data to display correctly

**Logs**  

```
14/07/2025 20:06:44 [ERROR] Error 401 fetching OpenWeatherMap API:
```

**Screenshots / photos of the Turing screen**  

[image]
[image]

**Environment:**  
 - Smart screen model : XuanFang 3.5”
 - Revision of this project  : 3.9.3
 - OS with version [e.g. Windows 11

Joespec556, 2025-07-18:
Same issue here with the 401 Error in the logs. The error on my display changes depending if it shows the beginning of the message or end of the 401 message. 

[image]

mathoudebine, 2025-08-09:
OpenWeatherMap has deprecated the old 2.5 API https://openweathermap.org/api/one-call-api
As it is stated in the Configure Window:
[image]

I will accept PR to use other services than OpenWeatherMap as I can understand people do now want to provide a payment method to OWM for a free API.