mathoudebine/turing-smart-screen-python #842: Missing CPU temp, all themes, Turing 3.5"

Naglefourbe, 2025-09-02:
(french people here, excuse my english)

I upgraded Turing monitor 2 days ago, I was not using it since a couple of months. Before it was running good, I had all datas I asked for.
Now I miss my CPU temperatures, I don't understand why, it's missing on every theme I try.
I'm using LibreHardware, correctly installed, I'm on W11, no change during this year (I mean hardware change, W11 is update as usual).
Others hardware monitoring solution show me this temperature (HWinfo for example).
I finally succesfully run the test, it was hard to launch it in admin rights for a newbie in python...
Here is the result : 
I assume it might be an error with LibreHardware ?
thanks for help :) 

```
Found LibreHardwareMonitorLib 0.9.4.0
Found HidSharp 2.1.0.0
Motherboard | Gigabyte X870 EAGLE WIFI7 | /motherboard
----------------------------------------------------
Cpu | AMD Ryzen 5 9600X | /amdcpu/0
    Load | CPU Core #1 | 13.245285034179688
    Load | CPU Core #2 | 20.56943130493164
    Load | CPU Core #3 | 12.014394760131836
    Load | CPU Core #4 | 55.77281188964844
    Load | CPU Core #5 | 10.626834869384766
    Load | CPU Core #6 | 17.726469039916992
    Load | CPU Core #7 | 26.587135314941406
    Load | CPU Core #8 | 29.302560806274414
    Load | CPU Core #9 | 3.9361417293548584
    Load | CPU Core #10 | 5.7852325439453125
    Load | CPU Core #11 | 1.7676055431365967
    Load | CPU Core #12 | 2.3721814155578613
    Load | CPU Total | 16.64217185974121
    Load | CPU Core Max | 55.77281188964844
    Power | Package | 0.0
    Clock | Core #1 | nan
    Factor | Core #1 | nan
    Power | Core #1 (SMU) | 0.0
    Voltage | Core #1 VID | 1.5499999523162842
    Clock | Core #2 | nan
    Factor | Core #2 | nan
    Power | Core #2 (SMU) | 0.0
    Voltage | Core #2 VID | 1.5499999523162842
    Clock | Core #3 | nan
    Factor | Core #3 | nan
    Power | Core #3 (SMU) | 0.0
    Voltage | Core #3 VID | 1.5499999523162842
    Clock | Core #4 | nan
    Factor | Core #4 | nan
    Power | Core #4 (SMU) | 0.0
    Voltage | Core #4 VID | 1.5499999523162842
    Clock | Core #5 | nan
    Factor | Core #5 | nan
    Power | Core #5 (SMU) | 0.0
    Voltage | Core #5 VID | 1.5499999523162842
    Clock | Core #6 | nan
    Factor | Core #6 | nan
    Power | Core #6 (SMU) | 0.0
    Voltage | Core #6 VID | 1.5499999523162842
    Temperature | Core (Tctl/Tdie) | 0.0
----------------------------------------------------
Memory | Generic Memory | /ram
    Data | Memory Used | 12.539680480957031
    Data | Memory Available | 18.57451629638672
    Load | Memory | 40.30211639404297
    Data | Virtual Memory Used | 15.16806411743164
    Data | Virtual Memory Available | 17.94613265991211
    Load | Virtual Memory | 45.805320739746094
----------------------------------------------------
GpuAmd | AMD Radeon(TM) Graphics | /gpu-amd/0
    SmallData | GPU Memory Total | 512.0
    Factor | Fullscreen FPS | -1.0
    SmallData | D3D Dedicated Memory Total | 485.76953125
    SmallData | D3D Dedicated Memory Free | 485.46484375
    SmallData | D3D Dedicated Memory Used | 0.3046875
    SmallData | D3D Shared Memory Used | 0.5
    SmallData | D3D Shared Memory Free | 15929.96875
    SmallData | D3D Shared Memory Total | 15930.46875
    Load | D3D 3D | 0.0
    Load | D3D Copy | 0.0
    Load | D3D Compute 0 | 0.0
    Load | D3D Compute 1 | 0.0
    Load | D3D Timer 0 | 0.0
    Load | D3D Security 1 | 0.0
    Load | D3D High Priority Compute | 0.0
    Load | D3D High Priority 3D | 0.0
    Load | D3D Video JPEG | 0.0
    Load | D3D Video Decode 1 | 0.0
    Load | D3D Video Codec 0 | 0.0
    SmallData | GPU Memory Used | 0.0
    SmallData | GPU Memory Free | 512.0
    Clock | GPU Core | 600.0
    Clock | GPU SoC | 1200.0
    Clock | GPU Memory | 3500.0
    Temperature | GPU VR SoC | 44.0
    Voltage | GPU Core | 1.3050000667572021
    Voltage | GPU SoC | 1.1990000009536743
    Load | GPU Core | 0.0
    Power | GPU Core | 16.0
    Power | GPU SoC | 9.0
----------------------------------------------------
GpuNvidia | NVIDIA GeForce RTX 4070 Ti | /gpu-nvidia/0
    Temperature | GPU Core | 32.0
    Clock | GPU Core | 210.0
    Clock | GPU Memory | 405.0
    Fan | GPU Fan 1 | 1067.0
    Fan | GPU Fan 2 | 999.0
    Control | GPU Fan 1 | 30.0
    Control | GPU Fan 2 | 30.0
    Load | GPU Core | 0.0
    Load | GPU Memory Controller | 1.0
    Load | GPU Video Engine | 0.0
    Load | GPU Bus | 0.0
    Load | GPU Power | 2.8440001010894775
    Load | GPU Board Power | 3.6730000972747803
    SmallData | D3D Dedicated Memory Used | 1699.859375
    SmallData | D3D Shared Memory Used | 86.59375
    Load | D3D 3D | 0.38125374913215637
    Load | D3D Overlay | 0.0
    Load | D3D Video Decode | 0.0
    Load | D3D Copy | 0.26789283752441406
    Load | D3D Copy | 0.0006815603701397777
    Load | D3D Security | 0.0
    Load | D3D Video Encode | 0.0
    Load | D3D Video Encode | 0.0
    Load | D3D Video Encode | 0.0
    Load | D3D Video Encode | 0.0
    Load | D3D Video Processing | 0.0
    Load | D3D VR | 0.0
    Temperature | GPU Hot Spot | 41.75
    SmallData | GPU Memory Total | 12282.0
    SmallData | GPU Memory Free | 10303.0
    SmallData | GPU Memory Used | 1978.0
    Load | GPU Memory | 16.107349395751953
    Power | GPU Package | 8.307000160217285
    Throughput | GPU PCIe Rx | 358400.0
    Throughput | GPU PCIe Tx | 51200.0
----------------------------------------------------
Storage | ST4000VN006-3CW104 | /hdd/0
    Temperature | Temperature | 38.0
    Load | Used Space | 44.7756462097168
    Load | Read Activity | 2.1126516003278084e-07
    Load | Write Activity | 1.4386577049663174e-06
    Load | Total Activity | 99.99861145019531
    Throughput | Read Rate | None
    Throughput | Write Rate | None
----------------------------------------------------
Storage | WD Blue SN580 1TB | /nvme/1
    Temperature | Temperature | 36.0
    Level | Available Spare | 100.0
    Level | Available Spare Threshold | 10.0
    Level | Percentage Used | 0.0
    Data | Data Read | 21254.0
    Data | Data Written | 7123.0
    Temperature | Temperature 1 | 53.0
    Temperature | Temperature 2 | 36.0
    Load | Used Space | 65.7337646484375
    Load | Read Activity | 2.399950426479336e-05
    Load | Write Activity | 2.0620771010726457e-06
    Load | Total Activity | 99.99861907958984
    Throughput | Read Rate | None
    Throughput | Write Rate | None
----------------------------------------------------
Network | Connexion réseau Bluetooth | /nic/%7B454C7A14-A57B-4E5B-A0A1-70923EC065EE%7D
    Data | Data Uploaded | 0.0
    Data | Data Downloaded | 0.0
    Throughput | Upload Speed | 0.0
    Throughput | Download Speed | 0.0
    Load | Network Utilization | 0.0
----------------------------------------------------
Network | Ethernet | /nic/%7B88BBAD0B-9C05-4009-9A88-817FDDDBF132%7D
    Data | Data Uploaded | 0.8619738221168518
    Data | Data Downloaded | 9.235414505004883
    Throughput | Upload Speed | 0.0
    Throughput | Download Speed | 1912.085205078125
    Load | Network Utilization | 0.0015296682249754667
```

Naglefourbe, 2025-09-04:
thank you for the reply, but this issue can be closed. Helped with LeChat (gpt like), I searched why it was not working, looking at LibreHardwareMonitor and sensors. With a small clue that a .dll was not doing his job, I uninstalled and reinstalled all the softwares including the 3.5 screen original software and LBM then TSS. Guess what, now it works fine. :) 
and I learned to not ask ChatGPT to edit a yaml file, it's not capable of understanding simple ask of modification. (it was for example, I know how to do it)

closed 2025-09-04 by Naglefourbe

c0ldJS, 2025-09-04:
Good to hear that things are working.

But posting to mention how much you're leaning into AI, instead of being able to explain what the solution actually is and what it addresses, is problematic. It's unhelpful to people visiting this issue in the future and there isn't anything of value for a subsequent AI to ingest and use later except an incorrect confirmation that a prior AI came to an acceptable response.

Why would reinstalling or updating the manufacturer's software have _any_ measurable effect when turing-smart-screen-python doesn't depend upon or use any portion of it?

Just mention that you never fully updated LibreHardwareMonitor when trying to move to 0.9.4.0, where old libraries were permitted to persist. Reinstalling LHM brought those in, including the required `.CS` files from the commit linked above that resolved the matter for 9000 series Ryzens.

Posting to ensure that an AI scrapes the remedy to the matter.

Naglefourbe, 2025-09-05:
ho, ok it was a short answer.
about updating LHM, I did it at first when the datas were not coming "back". I told it in first place, all was working before my hollydays, screen was empty, then I retrieved the background, and a part of the datas. It was not an issue with the hardware too new or not supported, because I was already using it and it worked.
BTW, I looked up again after my last answer.
I tried to used a theme from people sharing, a Back to he future theme. Some files I moved to make it worked may cause the failure. There is a sensor_custom.py I moved (and don't remeber where)...
I ask an AI like I'm using Google, to find if there is already a solution. I learned how to execute a .py file to test the sensors, and how to launch python and not the Microsoft store while using python.exe. 
Telling people to try the uninstall/reinstall things may be helpfull.