mathoudebine/turing-smart-screen-python #860: Got problem on 3.5 turing

KAC-PDW, 2025-09-24:
**Describe the bug**  
3.5 Turing with LHM option is not works.

**To Reproduce**  
turn on and Select Turing, 3.5, automatic, classic, 50%, [theme name], LibreHardwareMonitor...

**Expected behavior**  
my turing screen are turned on and display informatics.

**Logs**  
```
[PYI-3204:DEBUG] PyInstaller Bootloader 6.x
[PYI-3204:DEBUG] LOADER: argv[0]: C:\Users\Kabon Sirona\AppData\Local\Programs\Turing System Monitor\configure.exe
[PYI-3204:DEBUG] LOADER: executable file: C:\Users\Kabon Sirona\AppData\Local\Programs\Turing System Monitor\configure.exe
[PYI-3204:DEBUG] LOADER: trying to load executable-embedded archive...
[PYI-3204:DEBUG] LOADER: attempting to open archive C:\Users\Kabon Sirona\AppData\Local\Programs\Turing System Monitor\configure.exe
[PYI-3204:DEBUG] LOADER: cookie found at offset 0x7935B4
[PYI-3204:DEBUG] LOADER: archive file: C:\Users\Kabon Sirona\AppData\Local\Programs\Turing System Monitor\configure.exe
[PYI-3204:DEBUG] LOADER: application has onedir semantics...
[PYI-3204:DEBUG] LOADER: _PYI_ARCHIVE_FILE not defined...
[PYI-3204:DEBUG] LOADER: parent process level = -2
[PYI-3204:DEBUG] LOADER: process level = 1
[PYI-3204:DEBUG] LOADER: application's top-level directory: C:\Users\Kabon Sirona\AppData\Local\Programs\Turing System Monitor
[PYI-3204:DEBUG] LOADER: calling SetDllDirectoryW: C:\Users\Kabon Sirona\AppData\Local\Programs\Turing System Monitor
[PYI-3204:DEBUG] LOADER: splash screen is unavailable.
[PYI-3204:DEBUG] DYLIB: attempting to pre-load bundled copy of ucrtbase.dll: C:\Users\Kabon Sirona\AppData\Local\Programs\Turing System Monitor\ucrtbase.dll
[PYI-3204:DEBUG] DYLIB: loading Python DLL: C:\Users\Kabon Sirona\AppData\Local\Programs\Turing System Monitor\python313.dll
[PYI-3204:DEBUG] DYLIB: loaded Python shared library.
[PYI-3204:DEBUG] DYLIB: imported symbols from Python shared library.
[PYI-3204:DEBUG] LOADER: pre-initializing embedded python interpreter...
[PYI-3204:DEBUG] LOADER: using PEP-587 API...
[PYI-3204:DEBUG] LOADER: creating PyConfig structure...
[PYI-3204:DEBUG] LOADER: initializing interpreter configuration...
[PYI-3204:DEBUG] LOADER: setting program name...
[PYI-3204:DEBUG] LOADER: setting python home path...
[PYI-3204:DEBUG] LOADER: setting module search paths...
[PYI-3204:DEBUG] LOADER: setting sys.argv...
[PYI-3204:DEBUG] LOADER: applying run-time options...
[PYI-3204:DEBUG] LOADER: starting embedded python interpreter...
[PYI-3204:DEBUG] LOADER: setting sys._MEIPASS
[PYI-3204:DEBUG] LOADER: importing modules from PKG/CArchive
[PYI-3204:DEBUG] LOADER: extracted struct
[PYI-3204:DEBUG] LOADER: running unmarshalled code object for module struct...
[PYI-3204:DEBUG] LOADER: extracted pyimod01_archive
[PYI-3204:DEBUG] LOADER: running unmarshalled code object for module pyimod01_archive...
[PYI-3204:DEBUG] LOADER: extracted pyimod02_importers
[PYI-3204:DEBUG] LOADER: running unmarshalled code object for module pyimod02_importers...
[PYI-3204:DEBUG] LOADER: extracted pyimod03_ctypes
[PYI-3204:DEBUG] LOADER: running unmarshalled code object for module pyimod03_ctypes...
[PYI-3204:DEBUG] LOADER: extracted pyimod04_pywin32
[PYI-3204:DEBUG] LOADER: running unmarshalled code object for module pyimod04_pywin32...
[PYI-3204:DEBUG] LOADER: looking for PYZ archive TOC entry...
[PYI-3204:DEBUG] LOADER: path to PYZ archive stored into sys._pyinstaller_pyz...
[PYI-3204:DEBUG] LOADER: running pyiboot01_bootstrap.py
[PYI-3204:DEBUG] LOADER: running pyi_rth_inspect.py
[PYI-3204:DEBUG] LOADER: running pyi_rth_setuptools.py
[PYI-3204:DEBUG] LOADER: running pyi_rth_pkgutil.py
[PYI-3204:DEBUG] LOADER: running pyi_rth_multiprocessing.py
[PYI-3204:DEBUG] LOADER: running pyi_rth__tkinter.py
[PYI-3204:DEBUG] LOADER: running configure.py
[PYI-14756:DEBUG] PyInstaller Bootloader 6.x
[PYI-14756:DEBUG] LOADER: argv[0]: C:\Users\Kabo[n Sirona\AppData\Local\PrPograms\TYIuri-ng3204:D EBUGSystem] L MoOAnitor/main.exe
DER: OK.
[P[PYI-3204:DEBUG] LOADER:YI-14756:DEBUG] LOADER: executab manually flushing le file: C:\Users\Kabonst Sirona\AppData\Lodout cal\Programs\Turing System Monitor\main.exe
[PYI-14756:DEBUG] LOADER: trying to load executabland stderr...
e-embedded archive...
[PYI-3204:DEBUG] LOADER: cleaning up Python interpreter...
[PYI-14756:DEBUG] LOADER: attempting to open archive C:\Users\Kabon Sirona\AppData\Local\Programs\Turing System Monitor\main.exe
[PYI-14756:DEBUG] LOADER: cookie found at offset 0x7C7B81
[PYI-14756:DEBUG] LOADER: archive file: C:\Users\Kabon Sirona\AppData\Local\Programs\Turing System Monitor\main.exe
[... 55 lines cut ...]

경고: 어셈블리 바인딩 로깅이 꺼져 있습니다.
어셈블리 바인딩 오류 로깅 기능을 사용하려면 레지스트리 값 [HKLM\Software\Microsoft\Fusion!EnableLog] (DWORD)를 1로 설정 하십시오.
참고: 어셈블리 바인딩 오류 로깅 기능을 사용하도록 설정하면 그렇지 않은 경우보다 성능이 약간 떨어집니다.
이 기능을 끄려면 레지스트리 값 [HKLM\Software\Microsoft\Fusion!EnableLog]를 제거하십시오.

Please follow start guide to install required packages: https://github.com/mathoudebine/turing-smart-screen-python/wiki/System-monitor-:-how-to-start
Or the troubleshooting page: https://github.com/mathoudebine/turing-smart-screen-python/wiki/Troubleshooting#all-os-tkinter-dependency-not-installed
```

maybe seems RAMSPDToolkit-NDD are problem.

**Screenshots / photos of the Turing screen**  
Add screenshots or photos of the rendering on the smart screen to help explain your problem.
You can drag and drop photos here to add them to the description.

**Environment:**  
 - Smart screen model - Turing 3.5
 - Revision of this project - 3.9.4
 - OS with version - Windows 11
 - Python version - Python 3.10
 - Hardware Intel 13900H, 4070Laptop

**Additional context**  
Add any other context about the problem here.

abid112, 2025-09-24:
I've got the same problem. The latest version is not working for me. After installing the latest (3.9.4), after configuring Save and Run, nothing shows up on my screen.

GPleex, 2025-09-24:
Mine had the same problem, I used the .exe, went back to the previous version and it worked again

KAC-PDW, 2025-09-25:
Windows installer and manual installation version were had same problem on 3.9.4

mathoudebine, 2025-09-26:
Indeed the `RAMSPDToolkit-NDD.dll` file was missing. I removed the 3.9.4 release and created a branch https://github.com/mathoudebine/turing-smart-screen-python/compare/main...fix/lhm-pre-release
The issue is with this LHM pre-release version I lost the CPU frequency: values are always 0
So I may eventually change my mind and not commit the LHM pre-release if it brings regressions on sensors

ariltonJaguilar, 2025-10-01:
just reporting that i'm using the 3.9.3 version and my CPU temperature are always showing as 0