mathoudebine/turing-smart-screen-python #863: Support for 0.96 inch USB lcd displays

flow23, 2025-09-26:
**Is your feature request related to a problem? If so, please describe the problem.**  
No

**Describe the feature / solution to your problem you'd like**  
Support for 0.96 inch USB lcd displays

**Describe alternatives you've considered / and or tested**  
N/A

**Screenshots / photos & mockups of the Turing screen**  
N/A

**Environment:**  
N/A

**Additional context**  
[Link to product](https://s.click.aliexpress.com/e/_oB5rnbx)

mathoudebine, 2025-10-20:
@mhalano @flow23 can you test the branch `dev/863-support-for-weact-display-fs-096-inch-usb-lcd-displays` with your WeAct 0.96" display?
I've added support for both WeAct 0.96" and 3.5" displays, and 4 themes for the 0.96" model
I received my screens and tested on them, everything was working fine on my side!

mhalano, 2025-10-20:
It seems it can't find the fonts. There is a special place to put the fonts?
```
20/10/2025 20:06:10 [INFO] Loading theme Digital_CPU_0.96Inch from res/themes/Digital_CPU_0.96Inch/theme.yaml
20/10/2025 20:06:10 [DEBUG] Static COM port: /dev/ttyACM0
20/10/2025 20:06:10 [DEBUG] Using Python 3.13.7 (main, Aug 20 2025, 22:17:40) [GCC 15.2.0]

(main.py:1469589): libayatana-appindicator-WARNING **: 20:06:10.818: libayatana-appindicator is deprecated. Please use libayatana-appindicator-glib in newly written code.
20/10/2025 20:06:10 [INFO] Tray icon has been displayed
20/10/2025 20:06:10 [INFO] Initialize display
20/10/2025 20:06:11 [INFO] Device version: V1.0.0.0
20/10/2025 20:06:11 [DEBUG] Drawing Image: background
20/10/2025 20:06:11 [DEBUG] Bitmap /home/mhalano/code/turing-smart-screen-python/res/themes/Digital_CPU_0.96Inch/background.png is now loaded in the cache
20/10/2025 20:06:11 [DEBUG] Drawing Text: LOAD
Traceback (most recent call last):
  File "/home/mhalano/code/turing-smart-screen-python/main.py", line 209, in <module>
    display.display_static_text()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/home/mhalano/code/turing-smart-screen-python/library/display.py", line 153, in display_static_text
    self.lcd.DisplayText(
    ~~~~~~~~~~~~~~~~~~~~^
        text=config.THEME_DATA['static_text'][text].get("TEXT"),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<13 lines>...
        anchor=config.THEME_DATA['static_text'][text].get("ANCHOR", "lt"),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/home/mhalano/code/turing-smart-screen-python/library/lcd/lcd_comm.py", line 282, in DisplayText
    ttfont = self.open_font(font, font_size)
  File "/home/mhalano/code/turing-smart-screen-python/library/lcd/lcd_comm.py", line 693, in open_font
    self.font_cache[(name, size)] = ImageFont.truetype(name, size)
                                    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 880, in truetype
    return freetype(font)
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 877, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 285, in __init__
    self.font = core.getfont(
                ~~~~~~~~~~~~^
        font, size, index, encoding, layout_engine=layout_engine
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
OSError: cannot open resource
```
EDIT: Nevermind. The theme was using backslashes instead of slashes, and I'm on Linux. I could get the first theme, the simpler one, work.

mhalano, 2025-10-20:
@mathoudebine Can you make the path for the fonts use slashes the same way the path for the images? I tested the second theme and the path for the image was ok, but not the path for the fonts.